evansm7/pico-mac #4: pimorini vga demo support?

DuckyPolice, 2024-06-19:
im using the pimorini vga demo board and i am not getting output does this project support it? if not can you make a version that does? or maybe i built it wrong the build steps were confusing

evansm7, 2024-06-20:
Hello! No, not sure if you saw the readme but it only supports the wiring listed in the table there. Looking at the pimoroni board pinout, it’s different enough to require work (quite a change to the PIO code unfortunately).  It’s possible but then I don’t have a board to test with either, so I feel it’s unlikely I will get to implement this, but happy to give guidance if you want to have a crack at it.

zapewne, 2024-06-21:
Mobile mac with mechanical keyboard, meybe network and usb file as disk image?
And working time on one charge about week ;D

DuckyPolice, 2024-06-25:
the demo boards are around 30 bucks, they deliver fast, and are great. i use them for MEMU and rp2040-doom
its quite worth the extra work to impliment it. i only do python programming and mostly API and networking work so im not qualified to add it myself.

DuckyPolice, 2024-06-25:
but if you were able to provide me with dev builds, i would be happy to run/test them for you. my discord handle is "duckypolice" just incase you want eisier access to contact.

mix127, 2024-06-26:
some like this?
https://liliputing.com/olimex-neo6502pc-is-a-retro-pc-that-combines-an-apple-ii-compatible-chip-with-a-rp2040-microprocessor/

But with power and mechanical keyboard