evansm7/pico-mac #5: eject

zapewne, 2024-06-21:
What happened in emulator if # E
```
Opening ROM 'Mac-Plus.ROM'
RAM mapped at 0x7fc09c5b0000
Disc mapped at 0x7fc09c1a0000, size 819200
SonyControl:accRun: Not supported!
DISC: EJECT
```
on linux emulator, program crash

evansm7, 2024-06-26:
Not quite, it doesn’t crash; the eject event is hooked to (cleanly) exit the emulator.

closed 2024-06-26 by evansm7

fab1an2, 2024-06-26:
In my case, the program simply stops working. crashing

og1en, 2024-06-26:
People are making an effort. They are testing your solution. And you still resent them?

You discourage the next ones. After all, we all want this program to be useful.

evansm7, 2024-06-26:
> People are making an effort. They are testing your solution. And you still resent them?

Please don’t bring sass, we’re all doing this for fun. Sounds like you have misunderstood my reply, or something else – either way, be nice.

evansm7, 2024-06-26:
> In my case, the program simply stops working. crashing

Can you describe what’s happening if it’s not an exit? From previous description, the output you’d quoted looks normal (ie EJECT the last thing you see before an exit). Use detail. If an author can not recreate or see immediately from the issue report what’s happening, it’s unlikely they’ll be able to fix it.