evansm7/pico-mac #12: Compile error

TimRustige, 2024-08-22:
Hi, I get an error while compiling, see photo. Can you help please. I tried adding what could be a missing underscore to the video.c source and it then compiled, but I get no video out of the UF2 it's produced, although it's running something and the LED is flashing - whereas if I generate a UF2 file at picomac.bluescsi.com , that works correctly.
[image]

Are there some corrections made to the code on the bluescsi site that haven't made it into this tree yet?

Any help much appreciated.

evansm7, 2024-08-22:
Hi Tim -- I've just pushed a fix for this, thanks.  It was actually that a couple of source files missed a header which was included elsewhere in the `pico-sdk` previously (e.g. at v1.5.x), but no longer in v2.0.  Added 'em and it should be good to go.

TimRustige, 2024-08-23:
Thanks for this Matt. Got it running on a Pico2 now as well.