evansm7/pico-mac #13: SD Card disk image

horsiezoo, 2024-08-29:
Hi there, fantastic project! The first build is great, I deleted my ./pico-mac dir and started again this morning after I saw your update.
I'm having trouble understanding :
 "...copy your disc image into one of the following files in the root of the card:

umac0.img: A normal read/write disc image"

Do you mean create an empty .img say 10MB on FAT16 ? format SD card named umac0.img and then copy a known good mac boot disk image into that?

Or: rename a known good boot image to umac0.img on SD card ? 
(I've tried both)

Else if the above is correct, its my SD card wiring. 

Kind Regards !

horsiezoo, 2024-08-29:
Hi again nevermind, figured it out - yup SD wiring, luckily no magic smoke this time ! Keep up the great work ! 

Regards!

closed 2024-08-29 by horsiezoo

evansm7, 2024-08-29:
Ha wicked! Glad to hear it can be recreated 😁 have fun. 
(For any others wondering, any disc image file that works with umac will work, copied to SD and named umac0.img)