evansm7/pico-mac #14: Is it possible to enlarge the screen?

wskjinfen, 2024-09-05:
What a fun project, thanks to the author ^_^
Is it possible to enlarge the screen? It would be best if it could fill the entire monitor.

evansm7, 2024-12-13:
Actually yes...... there are two parts to doing this:

1. Hack the umac emulator to patch ROM to change screen size
2. Change pico-mac to alter resolution (640x480 should be easy, removing black borders)

I was stuck on (1) for a while, but I finally found all the places that needed patching.  I got the MacPlus ROM booting System 3, System 7 with different screen resolutions:

[image]

I will have to clean this up and merge there first, then import that into this project.  Give me time....

The other thing to bear in mind is big screen means less free RAM.  For 640x480, an extra 16KB is used.  For a true Mac 128K this is a bad idea; nothing will run.  But for a Mac 208K then that's reasonable, and with a Pico 2 / RP2350 there is enough memory.

So yes, but :-)  Watch this space.

evansm7, 2024-12-17:
https://github.com/evansm7/umac now supports varying the screen resolution.  I haven't modified pico-mac yet, but will try/update soonish. :)

evansm7, 2024-12-20:
Pushed v0.21: this now adds a build option of `-DUSE_VGA_RES=1`, which configures `umac` and the pico-umac video system to use 640x480.
Confirmed working on a Mac 208K configuration, on real HW (RP2350, but expect to work identically on RP2040).

closed 2024-12-20 by evansm7