evansm7/pico-mac #15: screen goes black when calling the WriteMacInt8/16/32 function

wskjinfen, 2024-09-08:
I want to add the ability to manually mount the disk via the GPIO switch after it is ejected (to achieve a process similar to the real 128k disk swap process). What function can I call to achieve this? In addition, I found that the screen will turn black after calling the WriteMacInt function. What is the reason for this? Thank you!

evansm7, 2024-09-08:
Gday… No idea. Black screen likely means a crash. 

I’d suggest working on umac (eg on Linux) and moving to the Pico once that works, as debugging on a Proper Computer would be much less frustrating. Have a look at BasiliskII’s disk driver, as that has some code that can mount a disk. From memory, it registers a timer callback which calls mount which drops into the driver… or something. I don’t really know how macOS works :)

evansm7, 2024-12-13:
Can I suggest we move this to umac first, there's a discussion here:  https://github.com/evansm7/umac/issues/10

Then we can open a to-do on pico-mac once that works.  Cheers!

closed 2024-12-13 by evansm7