evansm7/pico-mac #16: Midi (Feature Request)

Cyantig, 2024-10-26:
Would really love to see the addition of a Midi port. I am thinking that the pico mac would make a killer platform for running some vintage music production apps like Laurie Spiegel's Music Mouse.

Cyantig, 2024-11-21:
My bad. 

 I don't think this would work now that I am looking at it. I think a 512k was the model she started developing on on not the 128k.

probonopd, 2025-01-29:
Doesn't MIDI run over serial, and pico-mac does implement a serial port?
Maybe we might be able to get [MOTU Performer 1](https://macintoshgarden.org/apps/motu-performer-1) to run?

evansm7, 2025-01-30:
The physical serial port is for debug yep; the umac emulator doesn’t support serial IO (the SCC is a beast…). MIDI would be most interesting, but then there’s PPP/modem emulation (esp on a PicoW!), and bottom of my personal list (hardest) would be appletalk. 
Re MIDI, an unknown is whether the emulation is jittery or too slow for good MIDI, but worth a try sometime.

probonopd, 2025-02-05:
Upon further investigation, it appears that the Mac's MIDI implementation is more complex than initially thought. The standard baud rate of 31250 is not directly supported by the hardware. As a result, Macintosh MIDI adapters are more intricate and differ significantly from the typical passive serial-to-MIDI adapters.

evansm7, 2025-02-05:
Ha yes… there are various circuits and my SE/30 adapter generates a clock to inject. All that tho is thankfully a level of detail beyond simple SW functional emulation. But the SCC is still scary :) At least it’s plain serial (not HDLC/framed like appletalk).

evansm7, 2025-02-06:
Yes, the emulator provides a high level behaviour of the Mac - it doesn’t model anything like circuit-level detail, so to achieve serial IO umac would need to model the (async mode for) SCC serial regs/FIFOs/IRQs and ultimately capture (or provide) a stream of bytes from the host. On a PC that could come from eg a Linux MIDI API; on the Pico that would be interfaced to a HW serial port. 
The TX drive circuit, and RX optocoupler, would still be necessary.