evansm7/pico-mac #18: VGA Connection

Martius108, 2024-10-31:
Hello,

I want to solder half of an VGA cable to the Pico and I found this website for the VGA connector:

[https://pc-anwender.de/Blog/Blog-2010/04-07-vga-anschluss-pinbelegung.htm](url)

It is in german but I hope it is understandable. In contrast to your description in my case I have pins 1-3 RGB, 4 and 5 not connected, 6-8 GND, 10 (according to the above website) ground vertical synchronisation, 11 not connected, 12 SDA, 13 HSYNC, 14 VSYNC, 15 SCL. Is this correct? Is there no 5V connection between the Pico and the VGA device? 

Thanks 
Martin

evansm7, 2024-10-31:
Hello!
Just connect as many grounds as you have available (as many of 5-8,10 as are wired) all together to the nearest Pico Gnd. Maybe they’ve cheaped out and made the shield connect to all the ground pins? Anyway likely not critical as long as “some” are connected. :)
Then RGB 1-3 together, HS 13, VS 14, to the corresponding resistors. And then, don’t worry about any of the other pins (no I2C, no 5V). Your resistor values sound just fine, 68 will do. (If it doesn’t, try lower, but am pretty sure it’ll be fine.)
Viel Glück!

Martius108, 2024-11-04:
Works great! Thank you.

summerlotus513, 2024-11-04:
Hello, I also encountered a similar VGA problem
I set up and tested it at the beginning, I can work normally
Then I found the Universal Board to welded the components. I confirmed that my VGA 1-3 link was connected to the PICO 18 pins of PICO with 100Ω.
VGA 13 pin connect a 66Ω resistor (I get an approximate resistance value through two resistors) connected to the 21 pin of Pico
VGA 14 pin connect 66Ω resistance to the 19 pin of Pico
VGA's 5,6,7,8,10 pin I connect to the GND of Pico (the pin between 19 and 21)
Then he was strange that he didn't work!

summerlotus513, 2024-11-04:
> Hello, I also encountered a similar VGA problem I set up and tested it at the beginning, I can work normally Then I found the Universal Board to welded the components. I confirmed that my VGA 1-3 [...]

By the way, my display has been in the reciprocating signal and then lost signal.

Martius108, 2024-11-04:
Please be aware that GOIO 19 is not pin 19. See pinout of the Pico.

closed 2024-12-13 by evansm7