evansm7/pico-mac #19: VGA Library

Martius108, 2024-11-13:
Hi, 

I got this project running without problems. I hardwired the half of a VGA cable to a breadboard with the Pico. Now I’d like to make a new project - a Pong game. The code is not difficult, there are lot of examples out here. But I need a kind of VGA library. So my question is which part of this project is the VGA library and how can I implement it into the new project?

thanks
Martin

ventZl, 2024-11-13:
Note that there is a released/unreleased part of Pico SDK, which actually contains support for PIO bitbanged VGA and it is available here: https://github.com/raspberrypi/pico-extras/

There are even examples in pico playground repository so you can get the basic picture on how to use the library and it is even rather extensively documented (so you can understand how it works).

evansm7, 2024-12-13:
@Martius108 if you haven't gone the route suggested by @ventZl then yes it should be straightforward enough to reuse the video code elsewhere.  I'd wanted to do the same so designed it to be reusable.

In `include/video.h` you'll see one call:
`void    video_init(uint32_t *framebuffer);`

Include the following files in your project:

- include/video.h
- src/video.c
- src/pio_video.pio

In your CMakeLists.txt add `pico_generate_pio_header(firmware ${CMAKE_CURRENT_LIST_DIR}/src/pio_video.pio)`, and add  `hardware_dma` and `hardware_pio` to the `target_link_libraries( ... )` list.

Now, make a frame buffer somewhere:  a global `uint8_t framebuffer[512*384/8];` will do.  To start video, provide that address to the `video_init`, e.g. `video_init((uint32_t *)framebuffer);`.

The resolution is tuned to the Mac of course.  Try changing the `#define`s in `video.c`.  The way it is intended to work is it scans out 640x480 and then has a "visible" 512x342 with padding each side.  If you make the "visible" larger the padding _should_ just work automatically, even up to 640x480 -- I haven't tried this and there may well be bugs tho. 

The order of bits/bytes in the framebuffer is odd -- if you write a native program it'll be awkward.  It's that way because the Mac is big endian.  The DMA swaps bytes and the PIO shifts bits out top down.  So, try:

- In video.c `channel_config_set_bswap(&dc_tx_d, true);`  <-- change to false
- In pio_video.pio `sm_config_set_out_shift(&c, false /* OUT MSBs first */,  ...` <-- change to true

HTH!

evansm7, 2024-12-20:
> If you make the "visible" larger the padding should just work automatically, even up to 640x480 -- I haven't tried this and there may well be bugs tho.

Just to confirm it works out of the box (the latest pico-mac revision supports VGA resolution, and simply redefines `VIDEO_FB_HRES`/`VIDEO_FB_VRES` to 640x480.

Martius108, 2024-12-21:
Thank you. I will try this out.