evansm7/pico-mac #21: Use VGA setup for another project

Martius108, 2024-12-05:
Hello,

I successfully cerated this project and wanted to use this Pico - VGA hardware configuration for another project (Pong game). I tried to code this in MicroPython but could't get a VGA output. So I started to create a test file at first which looks like this - but still no signal on the screen. Could you please tell me what I'm doing wrong?

`from machine import Pin, mem32
import rp2
import array
import time

# Constants from pico-mac video.h
H_ACTIVE = 640
H_SYNC_START = 656      # H_ACTIVE + H_FRONT_PORCH (16)
H_SYNC_END = 752       # H_SYNC_START + H_SYNC (96)
H_TOTAL = 800         # H_SYNC_END + H_BACK_PORCH (48)

V_ACTIVE = 480
V_SYNC_START = 490     # V_ACTIVE + V_FRONT_PORCH (10)
V_SYNC_END = 492      # V_SYNC_START + V_SYNC (2)
V_TOTAL = 525        # V_SYNC_END + V_BACK_PORCH (33)

@rp2.asm_pio(
    out_init=(rp2.PIO.OUT_LOW,) * 1,    # Video starts LOW
    sideset_init=(rp2.PIO.OUT_HIGH,) * 2,  # Both syncs start HIGH
    out_shiftdir=rp2.PIO.SHIFT_RIGHT,
    autopull=True,
    pull_thresh=32,
    fifo_join=rp2.PIO.JOIN_TX
)
def vga_24mhz():
    # Based on pico-mac video.pio
    pull(block)              .side(0b11)  # Get first word of line
    out(x, 32)              .side(0b11)  # Initialize X with pixel count
    pull(block)             .side(0b11)  # Get next word ready
    
    label("bitloop")
    out(pins, 1)    [2]     .side(0b11)  # Output pixel data
    jmp(x_dec, "bitloop")   .side(0b11)
    
    # Front porch (16 cycles)
    nop()           [15]    .side(0b11)
    # HSYNC (96 cycles)
    nop()           [95]    .side(0b01)
    # Back porch (48 cycles)
    nop()           [47]    .side(0b11)

# Create a simple test pattern (white screen)
line_words = H_ACTIVE // 32
test_line = array.array('I', [0xFFFFFFFF] * line_words)

# Initialize pins
video_pin = Pin(18, Pin.OUT)  # GPIO18 -> Physical Pin 24
vsync_pin = Pin(19, Pin.OUT)  # GPIO19 -> Physical Pin 25
hsync_pin = Pin(21, Pin.OUT)  # GPIO21 -> Physical Pin 27

# Initialize state machine
sm = rp2.StateMachine(
    0,                    # State machine 0
    vga_24mhz,           # PIO program
    freq=25_175_000,     # Pixel clock frequency
    sideset_base=Pin(19),# VSYNC
    out_base=Pin(18),    # Video
    set_base=Pin(21),    # HSYNC
    out_shiftdir=rp2.PIO.SHIFT_RIGHT,
    autopull=True,
    pull_thresh=32
)

# Start the state machine
sm.active(1)

# VSYNC handling
vsync_counter = 0

print("VGA Test Pattern Started")

while True:
    # Handle vertical sync
    vsync_counter += 1
    if vsync_counter >= V_TOTAL:
        vsync_counter = 0
    
    # Set VSYNC signal
    if V_SYNC_START <= vsync_counter < V_SYNC_END:
        vsync_pin.value(0)  # VSYNC active (low)
    else:
        vsync_pin.value(1)  # VSYNC inactive (high)
    
    # Output test pattern during active video
    if vsync_counter < V_ACTIVE:
        for word in test_line:
            sm.put(word)
    
    time.sleep_us(31)  # Approximate line timing
`

evansm7, 2024-12-13:
No idea!  But it looks like you're doing timing-sensitive stuff in Python -- I'd be impressed if it was anywhere near fast/predictable enough to do that.  Even though C, it would be less frustrating to try this experiment in C I think... Good luck.

Martius108, 2024-12-21:
Thanks. That was a stupid idea :) I didn't know that this kind of project can only be done with C.

evansm7, 2024-12-21:
Ah but you don’t know until you try, experimenting is learning, and all that! (Who knows, maybe it’ll work :) but maybe worth trying it second :) :) )