evansm7/pico-mac #24: v0.21 boots to black screen

RonsCompVids, 2025-01-23:
After compiling v0.21, I am only able to get at black screen with a Happy Mac stuck in the middle.  If I power on with the SD card removed (SD contains an image bootable on previous builds), I will sometimes see a blinking question mark.

Firmware was compiled without a baked in disk image... but was configured for 640x480, SD and 208k support.

Any clued on what might be wrong?

Thank you!

[image]

probonopd, 2025-02-06:
@RonsCompVids I have built v0.21 with SD card support and a built-in disk image and I have never attached a SD card so far, and have never seen this issue.

Does it also happen for you if you remove all SD card circuitry? Possibly it thinks there is a SD card but doesn't find one for some strange reason?

If you like, you can also test with my [build](https://github.com/probonopd/pico-mac/releases/tag/continuous).

RonsCompVids, 2025-02-06:
I can certainly test it tonight.

I believe my error was not using the appropriate switched when compiling.

arulalan1, 2025-02-24:
Hi,
Did you resolve this ? I have exactly the same problem, very unhappily stuck with a happy Mac. I have a pico2 and am compiling for 410 kb ram and vga resolution.

closed 2025-02-25 by RonsCompVids