evansm7/pico-mac #42: Display problem

arulalan1, 2025-02-23:
I was able to compile Umac, and also now able to generate a UF2 file for my pico2 with 410 kb ram and vga resolution. However when I setup the hardware all I get is a small square with a happy Mac and a narrow strip at the bottom of the screen. 

Any ideas ?

TimRustige, 2025-03-02:
cmake .. -DPICO_BOARD=pico2

try it without VGA resolution, native works more consistently across different monitors for me.

TimRustige, 2025-03-03:
Keyboard & mouse should be attached before you power up. If you're using a shim-style micro USB to USB adapter, make sure it's inserted the right way up. Try a different keyboard & mouse.

TimRustige, 2025-03-04:
Try my build if you like https://www.tpr2.co.uk/128kSD.zip
I tried a wireless keyboard earlier without issues.

arulalan1, 2025-03-05:
Thanks Tim !
Will try it out and let you know.
A